FidoCadJ/FidoCadJ #184: user friendly selection dragging

PietroBaima, 2020-05-03:
When a tiny and dense area of the drawing is selected, and the user intends to move it, is not easy to click on an area not populated with those small red rectangles which allow the user to edit the drawing. That has the effect of cancelling the selection. 
I wonder whether is possible to provide a command (or a specific icon, dependending on usability) which abled the user working in a "move only" mode.
Thank you!

DarwinNE, 2020-05-03:
Very useful remark!
Maybe the best solution can be to introduce a specific command "Move" in the context menu that appears when right-clicking on the drawing.

PietroBaima, 2020-05-03:
Super duper! 
Thank you
I really appreciate your dedication with to this software indeed. No wonder it's so effective.