FidoCadJ/FidoCadJ #198: You require Java 9 on macOS but the latest officially supposed version is Java 8.

fxframes, 2020-08-02:
You require Java 9 on macOS but the latest officially supposed version is Java 8.
When trying to run FicoCadJ macOS says it's "broken" and can't open it.
You should not claim your software is compatible with macOS.

DarwinNE, 2020-08-02:
Hello @fxframes 
  thank you for your report.

The fact that the OS reports that the file is "broken" ("FidoCadJ.app is damaged and can’t be opened. You should move it to the Trash") is not due to an issue of the packet that is complete, but to the security settings of the operating system:

https://www.google.com/search?client=safari&rls=en&q=is+damaged+and+can’t+be+opened.+You+should+move+it+to+the+Trash&ie=UTF-8&oe=UTF-8

I understand the difficulty with the Java version, but my computer is a MacBook Pro and I use High Sierra as my main operating system and the Java I am using is version 14. The packaging system I am currently using is  jpackage, in theory it should create a self-sufficient package with all the dependences (hence the JRE):

https://openjdk.java.net/jeps/343

The packaged file was successfully tested on another Macintosh I have, but since it's an older machine, the security settings are probably less aggressive they tend to be on more recent versions.

I leave the issue open though, as other reported some difficulties, too: https://sourceforge.net/p/fidocadj/discussion/997486/thread/0c421064d2/

We are going to discuss here on GitHub, to see if there are solutions. Suggestions from the community are welcome, as usual.

Kind regards,
D.

DarwinNE, 2020-08-02:
If it can be useful, FidoCadJ 0.24.7 only requires Java 1.7 and can still be downloaded from there:

https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.7

It's a very stable version, that only contained some minor bugs and has been used on hundreds of different MacOSX machines. Of course, that will not solve the issue with "is damaged and can’t be opened..." message, as it is due to the security settings of the modern OS.

Regards,
D.

fxframes, 2020-08-03:
Hello @DarwinNE thanks for the quick reply.
I'll consider the alternatives.
It would be nice though to warn macOS users upfront that FidoCadJ won't work out of the box without some tweaking.
Best,

DarwinNE, 2023-01-19:
I tried downloading FidoCadJ 0.24.8 on a brand new MacBook Pro M2, running macOS Ventura 13.0.

At first the system asks to download and install Rosetta.
Then, the usual message "FidoCadJ.app is damaged and can’t be opened. You should move it to the Trash" is shown when one tries to launch the program, even after I moved it in the Applications folder. To fix this I did two things:
1 - Open a terminal and type xattr -c /Applications/FidoCadJ.app
2 - Right click on the executable and select "open". Then, the system asks me if I really want to run the software (downloaded on an untrusted source).

Hope it helps.

DarwinNE, 2023-05-19:
Here is a screenshot of the error.

[image]

JoopN, 2023-05-24:
> You require Java 9 on macOS but the latest officially supported version is Java 8. When trying to run FicoCadJ macOS says it's "broken" and can't open it. You should not claim your software is compatible with macOS.

Don't understand this, of course Apple want that you shop with them, but that is only for the interest of Apple and not for the user (I do have a very big paperweight with Mac logo). That is protecting own shop and not legal with present European rules. To be practical,  you can download OpenJDK version 11 on [https://bell-sw.com/pages/downloads/#/java-11-lts](url), also for Mac. If you want you can also download version 17.  Just take a look, your choice.

DarwinNE, 2023-05-24:
The FidoCadJ.app already ships with a full fledged version of Java. Having an error/warning message is normal (and important these days). The problem is that the error message is very misleading.

DanteCpp, 2024-09-23:
I have just tryed the xattr hack, but it doesn't seem to solve the problem. I still get the move to trash dialog window.

DarwinNE, 2024-09-23:
Hi @DanteCpp

If you go on the command line, put yourself where you put the FidoCadJ.app file and type `xattr FidoCadJ.app`, the `com.apple.quarantine` attribute is still there?