FidoCadJ/FidoCadJ #199: Thank you very much for the program under a free open source license!

mrkaban, 2020-08-08:
Thank you very much for the program under a free open source license! Added a program to a Russian-language site about free software, here is the [link](https://xn--90abhbolvbbfgb9aje4m.xn--p1ai/%D0%B3%D1%80%D0%B0%D1%84%D0%B8%D0%BA%D0%B0/%D1%81%D0%B0%D0%BF%D1%80/fidocadj-2d-%D0%B2%D0%B5%D0%BA%D1%82%D0%BE%D1%80%D0%BD%D1%8B%D0%B9-%D0%B3%D1%80%D0%B0%D1%84%D0%B8%D1%87%D0%B5%D1%81%D0%BA%D0%B8%D0%B9-%D1%80%D0%B5%D0%B4%D0%B0%D0%BA%D1%82%D0%BE%D1%80-%D0%B4%D0%BB%D1%8F-%D1%8D%D0%BB%D0%B5%D0%BA%D1%82%D1%80%D0%BE%D0%BD%D0%B8%D0%BA%D0%B8.html).

Do you plan to translate into other languages? I'm ready to try with translation into Russian. I am not a professional translator and professional radio engineer, but I am ready to give it a try.

DarwinNE, 2020-08-09:
Hello @mrkaban 
it would be great if you could translate FidoCadJ in Russian!

I wrote some instructions about how to translate the user interface back when the project was hosted on Sourceforge: https://sourceforge.net/p/fidocadj/discussion/997486/thread/9d08d94d/

If it is ok for you, you can create a new message bundle file with the Russian resources and do a pull request here on GitHub.

Since sometimes, during the development, it happens that language resources are added, I keep an issue for each translation (if it is OK for you, I will rename this one). Here is what we did for the other languages:

https://github.com/DarwinNE/FidoCadJ/issues?q=label%3Atranslation+

If an issue is closed, this means that the translation is complete.

Cheers,
D.

mrkaban, 2020-08-10:
I attach a file with translation
[MessagesBundle_ru.zip](https://github.com/DarwinNE/FidoCadJ/files/5051627/MessagesBundle_ru.zip)

DarwinNE, 2020-08-10:
Hello @mrkaban 
Thank you very much, but it would be much better to fork the project, add the file and do a pull request.
I will review it and accept it if there are no issues.

Cheers,
D.

DarwinNE, 2025-12-27:
I'm sorry but I forgot about this issue. I added the Russian translation of the UI in the commit [86144e0](https://github.com/FidoCadJ/FidoCadJ/commit/86144e03c962ae63669d04c06d89a904c641a4ea)

There are some resources that remain untranslated, as they were added in the last few years:
~~~~
inclusive_selection=Inclusive selection
completive_selection=Selection of complete elements
Theme_management=Theme and color management
Theme_groupBox=Theme management
Theme_enableSupport=Enable support for themes
FlatLaf_dark=Dark theme
FlatLaf_light=Light theme
FlatLaf_custom=Customize
Colors_management=Color management
Grid_dots_color=Grid color (dots)
Grid_lines_color=Grid color (lines)
Select_color=Selected elements color
Circuit_backgroud=Drawing area background color
Primitive_color=Default drawing color
Select_RL_color=Right-to-left selection rectangle color
Select_LR_color=Left-to-right selection rectangle color
Custom_theme_info=[The custom theme extends one of the two base themes (dark or light)]
alignLeftSelected=Align left
alignRightSelected=Align right
alignTopSelected=Align top
alignBottomSelected=Align bottom
alignHorizontalCenterSelected=Align horizontal center
alignVerticalCenterSelected=Align vertical center
alignmentMenu=Alignment
distributeHorizontallySelected=Distribute horizontally
distributeVerticallySelected=Distribute vertically
GhostPrimitivesFound=Some elements have been drawn outside the drawing area.
GhostPrimitivesTranslatePrompt=Do you want to translate all the elements to bring them into the drawing area?
GhostPrimitivesTitle=Hidden elements found

Zoom_key=Use the CTRL key to zoom
Move=Move
New_versin=A new version of FidoCadJ is available!
Current_version=Current version:
Latest_version=Latest version
Open_webpage=Do you want to open the download page?
Update_Available=Update Available
Please_visit=Please visit:
Open_Browser=Open Browser
~~~~