FidoCadJ/FidoCadJ #203: Update site darwinne.github.io/FidoCadJ/

gremirarunico, 2021-06-19:
Hi @DarwinNE
As we discussed on electroyou I am writing to suggest you a site update. Also all style sheets are broken due to github's update from http to https.

What I propose:
* Use HUGO as static CMS https://gohugo.io/
* Use hugo-book as template https://themes.gohugo.io/hugo-book/
Here you can fine a preview of what I have already done.
https://streamable.com/qgduk7

Why using Hugo? Because you can update pages as Markdown files. The template is automatically updated. It's easy managing multi-language and you can create "global vars" to use in all your documents.

DarwinNE, 2021-06-19:
Hello @gremirarunico 
Nice to see you there!

As I wrote on Electroyou (reminder: https://www.electroyou.it/forum/viewtopic.php?f=4&t=81997 ), I don't like very much to work on the website. It's ok for me to update the site using more modern techniques. Do you think you may follow the updates (not frequent) for the next two years?

If yes, the FidoCadJ website is contained in the gh-pages branch of the FidoCadJ project: 

https://github.com/DarwinNE/FidoCadJ/tree/gh-pages

You can fork the project, update that branch and then do a pull request.

Cheers,
D.

DarwinNE, 2021-06-19:
BTW, there are some open issues tagged "website":

https://github.com/DarwinNE/FidoCadJ/labels/website

It may be a good occasion to see if they are still relevant with the new version of the website and close them down if they are fixed.

Cheers,
D.

gremirarunico, 2021-06-19:
> Do you think you may follow the updates (not frequent) for the next two years?
Yes I can.
> It may be a good occasion to see if they are still relevant with the new version of the website and close them down if they are fixed.
All mobile-friendly requests will be solved.

Also I think I can use a directory for the webroot and you can set it as webroot in the project config page. In this way is easier to keep separate the building code and the website.

DarwinNE, 2021-06-19:
That sounds really interesting, but where am I supposed to set the webroot? Is there an option in GitHub?

gremirarunico, 2021-06-20:
Yes, go Settings (of the project), then pages and you find it ;)

DarwinNE, 2021-06-20:
Ok, I think I found it.

gremirarunico, 2021-06-20:
Can you review the english version?
Until it isn't approved I don't add translations (so the buttons are broken and you land to an empty page).
https://gremirarunico.github.io/FidoCadJ/

DarwinNE, 2021-06-20:
Hi, @gremirarunico 
Terrific job!

Very nice, fast and clean! I had a quick look at the page source and I did not notice anything too strange.
Here are some minor points. I also indicate some updates to the website that may be done with respect to the current version. They can be addressed in a second moment, if it is relevant.

https://gremirarunico.github.io/FidoCadJ/
- Can we find a title shorter than "FidoCadJ a free graphical editor for (more than) electronics"? Maybe just show "FidoCadJ", with a subtitle "a free graphical editor for (more than) electronics" or maybe (since nobody is using FidoCadJ right now besides electronics) just "a free graphical editor for electronics"
- I would use "Features" instead of "Right" as the title of the second column

https://gremirarunico.github.io/FidoCadJ/docs/screenshots/
- Change the title into "Screenshots"
- Is it possible to enlarge the picture when it is clicked?

https://gremirarunico.github.io/FidoCadJ/docs/download/
- Nice, very clear!
- "Currently under development" can become "Participate to the development"
- The Android version must be put below (old version or something), as its development has been abandoned.
- "Old stable version" may be updated to version 0.24.7 that only requires Java 7.
- Review the titles of the right sidebar.
- Replace the MacOSX logo with the macOS one, having a transparent background.
- Update the link to get Java.

https://gremirarunico.github.io/FidoCadJ/docs/code/
- Maybe the code of the second example is too long, it can be shown only if requested?
- Would it be interesting to add other examples?

https://gremirarunico.github.io/FidoCadJ/docs/libraries/
- Maybe reduce the text size in the table? I have to scroll when I look at it on my laptop (small screen).
- Maybe center text in the second and third column?
- I think I haven't yet added the library described here: https://www.electroyou.it/nicsergio/wiki/libreria-fidocadj-simboli-elettrici-per-schemi-funzionali (the author explicitly agrees in the comments)

https://gremirarunico.github.io/FidoCadJ/docs/faqs/
- Terrific!
- I think we should add a question about how to run FidoCadJ on macOS, the message "FidoCadJ.app is damaged and can’t be opened. You should move it to the Trash" unfortunately appears almost always with people using the last OS versions :-( More info here: https://github.com/DarwinNE/FidoCadJ/issues/198
- Another very frequent q&a to add is about Java 8 vs Java 9, many automatic updates are stuck to Java 8 :-(

https://gremirarunico.github.io/FidoCadJ/posts/
- Nice idea to add this page!

In general:
- The look is very sober and effective, but maybe a little too blunt. Maybe a touch of colour can be added here and there (for instance the titles of the pages can be made blue).
- The language selection will appear to the bottom of the pages? If so it may not be a good idea: people would immediately turn their attention elsewhere if they don't find the language button immediately. A good spot would be immediately above or below the menu on the left.
- Most of the times, people do not understand for what FidoCadJ is really useful. Look at the comments here: https://hackaday.com/2020/07/26/new-release-of-fidocadj-draws-schematics-everywhere/#comments The site may contain somewhere information about why it is still relevant to use FidoCadJ even if you have KiCad (or even Cadence) installed on your computer.

gremirarunico, 2021-06-21:
Thank you a lot!
> I would use "Features" instead of "Right" as the title of the second column
Ooops

> "Old stable version" may be updated to version 0.24.7 that only requires Java 7.
Ok, but for that version I think there isn't a manual. I use 0.24.8?

> Maybe center text in the second and third column?
What about removing completely the Version Column?

> The look is very sober and effective, but maybe a little too blunt. Maybe a touch of colour can be added here and there (for instance the titles of the pages can be made blue).
It can be changed changing css. But I realized whenever I change something in graphics I have the ability to make everything ugly and kinky. The template is designed for documentation. I could try but I would leave it as the last thing to do 

> Update the link to get Java.
The link is updated. For MacOS last official java release is the 8. But no problems with .app.

I done everything except adding clickable images (I need update the template. Something like this is ok? https://medium-zoom.francoischalifour.com/)
And collapsing text in code. I think it is useful only if we add more examples.

DarwinNE, 2021-06-21:
> Ok, but for that version I think there isn't a manual. I use 0.24.8?

I think it would be better to use the manual of the version 0.24.6: https://github.com/DarwinNE/FidoCadJ/releases/download/v0.24.6/manual_en.pdf
 
> [about library available on the website]
> 
> What about removing completely the Version Column?

I'd like to, but how to tell that a library has been updated? It has happened in the past. Would be enough to write this information in the description column? Is it possible to write the size in the first column, too? For instance, LaddEYr (zip, 2 KiB)

> [changing titles]
> It can be done changing css. But I realized whenever I change something in graphics I have the ability to make everything ugly and kinky. The template is designed for documentation. I could try but I would leave it as the last thing to do

Ok.

> 
> > Update the link to get Java.
> 
> The link is updated. For MacOS last official java release is the 8. But no problems with .app.

No, there are more recent releases. See for instance: https://jdk.java.net/16/
The .app contains a full Java 14 runtime environment. This has apparently became common (and is not a bad idea in itself). The issue with macOS are the security settings that in many cases make it very difficult to run apps downloaded outside the official store :(

Probably the link to give for a generic place where to find an up-to-date version of Java is this one:
https://openjdk.java.net

> I done everything except adding clickable images (I need update the template. Something like this is ok? https://medium-zoom.francoischalifour.com/)
> And collapsing text in code. I think it is useful only if we add more examples.

It's nice, but if it is complicated or you need to use other libraries, leave them as they are. A simple old-fashioned  alternative can be to add a link to the thumbnail pointing towards a full-size picture, but I don't know if it is applicable there.

gremirarunico, 2021-06-21:
> Would be enough to write this information in the description column?

I compressed the version as YY.M. On my screen it is solved.

> It's nice, but if it is complicated or you need to use other libraries, leave them as they are. A simple old-fashioned alternative can be to add a link to the thumbnail pointing towards a full-size picture, but I don't know if it is applicable there.

I used a dummy workround (another page with a link for the homepage).

Also I added "spoiler" in the code page.

I think more or less everything is done. Except for updating the documentation, but I think that is should be done in another session (and is a good thing that more than one person work on it).
What about this link https://sourceforge.net/p/fidocadj/discussion/?source=navbar ? It is active?

DarwinNE, 2021-06-21:
Great work, as usual. Here are a few remarks:

- for the version 0.24.7, the required Java version is 1.7 (a.k.a Java 7), not 1.5.
- in the download page, "Participate to the development" can be put at the same level of "Free download stable"
- change "A preliminary version of FidoCadJ for Android™ (4.0 at least). Almost stable, any feedback will be welcomed." into "An old preliminary version of FidoCadJ for Android™ (4.0 at least)." since the Android version is no longer developed.
- maybe a table is no longer necessary in the the "participate to the development" section.
- I am wondering if the "size" column is really useful altogether in the libraries page. What does it mean KN, by the way?

Cheers,
D.

DarwinNE, 2021-06-21:
I almost forgot...

> What about this link https://sourceforge.net/p/fidocadj/discussion/?source=navbar ? It is active?

More or less, I still receive some questions from time to time on the Sourceforge forums. People seem to be less scared to write there than on GitHub. I also receive questions in the Italian language, there.

gremirarunico, 2021-06-22:
> I am wondering if the "size" column is really useful altogether in the libraries page. What does it mean KN, by the way?

I think it isn't everywhere and I needed to update a lot of sizes because they wasn't updated. It was a typo (I wanted to replace KiB to KB (that is is again wrong, now it is corrected to kB)).

> More or less, I still receive some questions from time to time on the Sourceforge forums. People seem to be less scared to write there than on GitHub. I also receive questions in the Italian language, there.

Ok so I leave it? Perhaps may be a good think to completely close sourceforge repo and use only github.

Nevermind. Can now I proceed with the translations? Than we could think for some improvements (for example I think it will be very cool to publish directly the manual on the site in HTML and don't use only the PDF format).

DarwinNE, 2021-06-22:
> I think it isn't everywhere and I needed to update a lot of sizes because they wasn't updated. It was a typo (I wanted to replace KiB to KB (that is is again wrong, now it is corrected to kB)).

KiB was not an error: https://en.wikipedia.org/wiki/Kilobyte      :-)

> Ok so I leave it? Perhaps may be a good think to completely close sourceforge repo and use only github.

Don't leave the link. However, I will not close the sourceforge project, as FidoCadJ is still frequently downloaded from there (~500 downloads/month).

> Nevermind. Can now I proceed with the translations? Than we could think for some improvements (for example I think it will be very cool to publish directly the manual on the site in HTML and don't use only the PDF format).

It can be interesting to provide the manual as HTML, I may have a look at LaTeX2HTML, if it is simple enough to use.

gremirarunico, 2021-06-23:
> KiB was not an error: https://en.wikipedia.org/wiki/Kilobyte :-)

Yes I know, but KB isn't, and Firefox give the sizes in kB not in KiB.

> Yes, you can. It can be interesting to provide the manual as HTML, I may have a look at LaTeX2HTML, if it is simple enough to use.

Better this one https://tex.stackexchange.com/questions/341899/latex-to-markdown-converter and it should be very easy.

If for you is ok I'll require a pull request https://gremirarunico.github.io/FidoCadJ/ and I'll work to the greek translation more calmly because for me is a bit difficult to check everything.

DarwinNE, 2021-06-23:
> Yes I know, but KB isn't, and Firefox give the sizes in kB not in KiB.

Go for kB if you prefer. I don't think the difference matters, here.
 
> > Yes, you can. It can be interesting to provide the manual as HTML, I may have a look at LaTeX2HTML, if it is simple enough to use.
> 
> Better this one https://tex.stackexchange.com/questions/341899/latex-to-markdown-converter and it should be very easy.

I have had a look at LaTeX2HTML and it spent half a day to install (compiling 3 different versions of gcc, several versions of Python, etc...)

I managed to succeed and I tried it, but the result is not very good out of the box. I'll have a look at the alternatives.

> If for you is ok I'll require a pull request https://gremirarunico.github.io/FidoCadJ/ and I'll work to the greek translation more calmly because for me is a bit difficult to check everything.

Ok, go for it!

gremirarunico, 2021-06-23:
> Go for kB if you prefer. I don't think the difference matters, here.

I am a SI lover with some mental problems ;)

What about removing https://github.com/DarwinNE/FidoCadJ/tree/gh-pages/icons ? Are they useful?

> I have had a look at LaTeX2HTML and it spent half a day to install (compiling 3 different versions of gcc, several versions of Python, etc...)

I know, it's a very boring matter :(

DarwinNE, 2021-06-23:
> What about removing https://github.com/DarwinNE/FidoCadJ/tree/gh-pages/icons ? Are they useful?

Yes, they are, they are the original drawings done with Gimp of the icons.

DarwinNE, 2021-06-23:
Merged #204!

It seems to works fine. Great job!

closed 2021-06-25 by gremirarunico

DarwinNE, 2023-01-19:
Hello,
   I reopened again this issue as I think we need a small update to the FAQ version of the FidoCadJ website. Would it be possible to add a question related to running FidoCadJ on macOS systems? I gave it a try and it is cumbersome, yet still entirely possible: https://github.com/DarwinNE/FidoCadJ/issues/198

DarwinNE, 2023-01-19:
Hello,
   I reopened again this issue as I think we need a small update to the FAQ version of the FidoCadJ website. Would it be possible to add a question related to running FidoCadJ on macOS systems? I gave it a try and it is cumbersome, yet still entirely possible: https://github.com/DarwinNE/FidoCadJ/issues/198

Cheers,
D.

DarwinNE, 2023-05-19:
I finally preferred to prepare a small readme file distributed along the macOS application. I think it should make it more visible.