FidoCadJ/FidoCadJ #210: Change component color

Max2433BO, 2021-12-11:
Hi @DarwinNE 

I leave you this reminder of a request made by the user _gianniniivo_ of electroyou.it:

It could be implemented, in the new version of FidoCadJ, the possibility of changing the color of the components as is currently possible for the other elements (lines, curves, letters, ...)?

Bye, Max

JoopN, 2022-05-24:
??? That is already possible. Mark a line, go to the bar with the drop down menu which is probably on "circuit", open it and choose a color. You can do this too with complex parts from the lib. Place the part in the workarea, de-vectorize the part and give a component of that part a different color in the same way as above. Of course its only for that particularly drawing in which you changes it. It won't work in an other drawing, that has to do with compatibility.

JoopN, 2022-05-25:
No, you don't loose the symbol, its still available in the lib. Only in your drawing its changed, all symbols with ability of colour can change this way, other components need to be de-vectorized and can change then from color. And its just only in your drawing, next drawing is again the standard.

DarwinNE, 2022-05-26:
Of course you do not loose the symbol in the library. What I meant is that you have the coloured symbol that is not anymore a symbol (a macro in the FidoCadJ jargon). That means that if you try to move it, you have to move all the primitives. 

I think this is the issue that @Max2433BO is talking about.

JoopN, 2022-06-13:
I want to show you what I have in my own lib up to now  
[image]

For me this is what I want with color, as an example, a led emits light, but the component is the same as all others with different light, so why make the body of the diode in a different color?

DarwinNE, 2022-06-25:
It may make sense when, for a reason or another, you want to focus on a certain part of your circuit:

[image]

However, symbols contain their own elements and each one has its layer. Therefore, one should implement a technique to override it and this is not something I am planning to do, at least in a near future.

JoopN, 2022-06-25:
I know,  you wrote something about it, I just want to show how I solved the problem for a part, however my Italian is absent so I don't know exactly what gianniniivo of electroyou.it wants.