FidoCadJ/FidoCadJ #211: Gridpoint visible

JoopN, 2022-04-24:
When I'm designing a new item for my lib then I'm starting with something from the other libs or from scratch. It would be a great help if you can see on what grid point your mouse is on. Now you get with right mouse button 0.0 from the ruler. Could there be an option that you get the grid point your mouse is on?

DarwinNE, 2022-04-24:
Hello,
  I am not sure to understand what you mean. The coordinates that are shown next to the toolbar (the exact location may depend on the operating system you are using) is not enough?

[image]

Regards,
D.

JoopN, 2022-04-25:
True, but I use a pen on a tablet, so no fixed things and just set your eyes to the bar for the coordinates have the effect that you will move a little when on mm grid. With a big screen those coordinates are far away so to speak. When focussed on the spot you could click pen button (right click mouse) without moving. Made some rotary switches and setting the contacts right was sometimes one mm down or up or left or right in order to get the right picture. I made my lib in a text editor, not in FidoCadj. So FidoCadj open with object under construction and text editor next to it for adjusting and recording the coordinates on a Raspberry Pi 400 with OpenJDK 11.

JoopN, 2022-06-21:
The coordinates do not update when you drag a line over your drawing area. Only when you stop its updated, but that is very annoying in the drawing process.

DarwinNE, 2022-06-25:
I could reproduce the latter issue, indeed, it would be useful to update the information shown.