FidoCadJ/FidoCadJ #212: Manuals

JoopN, 2022-04-24:
If I'm going to translate the manual then its not only the manual which has to be translated but also all pictures if there is a part of the menu visible. How are we going to do that? Taking a screenshot is not a problem if you can use  the source of the picture, ie  download the fcd file, load it into the nationalised version of FidoCadj and make a screenshot. But now the problem where to put in on Github?

DarwinNE, 2022-04-24:
Hello,
   thank you for your intention to translate the manual! Which language are you targeting, in particular?

When I worked on the French and Italian manuals (that now are unfortunately outdated), I used the same pictures and screenshots I took for the English version of the manual.
Of course, it would be better to grab new screenshots.

Every resource related to the manual is present here:

https://github.com/DarwinNE/FidoCadJ/tree/master/manual

If you think you will add pictures associated with your translation, you may create a sub-directory of "images".

By the way, I received very few feedbacks about the english version of the manual, the only one I currently maintain. If you spot paragraphs that are not clear or that give obsolete information, do not hesitate to open issues to discuss about them.

Cheers

JoopN, 2022-04-25:
I want to translate it to Dutch, don't know how long it will take. Did translate jobs before and it took a few months up to two years for a large manual (100+ pages) with also translating the program. FidoCadj is already translated to Dutch, so it should be an easy task.

JoopN, 2022-04-26:
I downloaded the English manual, a LaTex file. What program do you use for maintaining this file? I have jEdit, but I'm not pleased with this package for this job.

DarwinNE, 2022-05-01:
I use a standard text editor, with LaTeX syntax highlighting. Textwrangler on macOS, to be more precise. I have an old MacBook Pro as my primary computer. The day I will change my computer I will check BBEdit.
On Linux I sometimes use Vim, but I have some troubles memorizing all the shortcuts and commands. I think I might use Vim if I had to develop more often on Linux.

JoopN, 2022-05-02:
Thanks, Vim got me moving with a search in the browser. Didn't succeed there, but got a top 12 list for LaTex editors. I'm on a Raspberry Pi so I settled for TEXstudio and I tried a part with translation, so now I have a FidoCadj manual partly English and  partly Dutch. Lets see where this end ;-)

JoopN, 2022-05-21:
Update.. I am at chapter one, before you think nothing happened, there is a lot which goes before chapter one. In chapter one I had to make some decisions about pictures with examples of old versions. Those have French or Italian in the bar and I decided not to redo it to Dutch. All pictures with actual version are redone with Dutch layout. Doing so I remarked that the translation of Dutch in the program is a little off, so I have to redo that too. Also I have to learn Latex along I proceed in the manual.

DarwinNE, 2022-05-21:
That's some very good news!

BTW could you correct the small error about \pdfbookmark in the English version of the manual?

JoopN, 2022-05-22:
Yes I can correct the small "errors" in the English version of the manual, but first I want to complete the Dutch translation, after that corrections and may be insert some new parts. With the translation I get deeper and deeper into the program, also I'm busy  with designing new electronic test equipment and found out that my math for electronics is a little rusty after 20+ years absence. The problems of 20+ years back are still there, no progress which amazes me.

JoopN, 2022-05-26:
I am somewhere in chapter 2, but I got lost in the bars, probably something from old versions. Made pictures en/nl with names how the bars are called. Have to see what happen if pdf is made and if it is in the right place. After that insert it into the English manual.

JoopN, 2022-06-06:
You(?) wrote this in the manual (2.1 second part, page 6)
"Such changes in the settings become effective after a restart of the
application, since it is arguably something that we may want to change
every day." 

I think you forgot the word "not" between "since it  is" and "arguably something", because I have a hard time in believing that  users are switching every day or every month or so. This is in my mind once set never touch again. But who knows, may be I have it wrong.

JoopN, 2022-06-06:
I want to insert this almost at the beginning of P2;
[image]

With it some naming what refers to what, because in the manual some are mixed up.

DarwinNE, 2022-06-07:
> "Such changes in the settings become effective after a restart of the
> application, since it is arguably something that we may want to change
> every day."

Oh, yes, the "not" is definitely missing in the sentence!

JoopN, 2022-06-09:
I had made the picture for adding to the English version, but the picture might not be a definitive version, just for me as a reference. I have to do some upgrading in the English manual, this will come along, have to fix some text too. I haven't looked at  the setup, don't know if I can change looks in Linux, not important for me at the moment. Things do differ between Apple and the rest of computers, so I don't know if you can keep it to one look, think you have to make more looks.

JoopN, 2022-06-23:
I'm constant falling over this error:
File `pgf.sty' not found. \usepackage
What is it, something special in Apple? Where can I find it, now I can't make pdf or export to html and I want to see how it is going in order to set some things right in the end product, ie tables and pictures on the right place.

DarwinNE, 2022-06-25:
> I'm constant falling over this error:
> File `pgf.sty' not found. \usepackage

You should have TiKZ/PGF installed in your LaTeX distribution:

https://www.ctan.org/pkg/pgf

It is a pretty common package, it is much probably part of your distribution, but you have to see how to install it. IIRC it may not be included in some "base" install.

> I think it is usefull to make a chapter with how to do with examples and how you get there. What' s your opinion on this?

What do you mean, exactly?

JoopN, 2022-06-25:
I think it is usefull to make a chapter with how to do with examples and how you get there. What' s your opinion on this?

What do you mean, exactly?
Somewhere there is a comment on more examples. Might be interesting to show how to get to something , how to make your own lib, within FidoCadJ or with an ASCII editor.  How to start a schematic drawing, add colour to it. Make colours specific for that drawing, next drawing will be again the standard. Show a complete small design. Do something totally different like my setup for furniture for a room. That kind of stuff.

JoopN, 2022-06-25:
Op Sat, 25 Jun 2022 14:36:23 -0700
Davide Bucci ***@***.***> schreef:

> You should have TiKZ/PGF installed in your LaTeX distribution:
>
> https://www.ctan.org/pkg/pgf
>
> It is a pretty common package, it is much probably part of your
> distribution, but you have to see how to [...]
Yes, its even installed, only its called probably differently, so
I have to adjust TexStudio. Thanks, I now now where to look for.

JoopN, 2022-11-19:
Still had problems. Today I decided to use your English manual and that one DOES compile, so that's good news. This means that all dependencies are there on my system. Now I have to look for what's wrong in my translation (so far somewhere chapter two), may be I translated a word too much ;-)

JoopN, 2023-05-09:
Sorry for some delay, was work related and it took me 3 months to solve problems. Between I did more tests and the problem was my approach of translating, ie some parts here, some parts there with still the English parts present. All has to do that a strait translation gives "broken" Dutch, but not the way we use it daily. I saved the file for copy/past  already translated parts. At some point I deleted all and copy back the English part. So that gives a document part Dutch part English. The good news then is that this document DOES compile. Way to go now is to translate a part and try to compile it, if success go on if error try to find out what did go wrong. I think that this approach will speed up things.

JoopN, 2023-05-12:
I'm almost at chapter 3... Lots of fights with Latex in order to get things where I wanted it, not full 100%, but satisfying enough. An other thing which doesn't go at the moment is to change the locale from English to Dutch in the document. Something with Babel, but where it comes from in my system I don't know. Linux isn't friendly in locating files, even so with search.

JoopN, 2023-05-13:
Okay, I have now a manual in Dutch, say version 0. How to proceed? I want that you take a look at  it because I changed a few things. One of those things is that  issue 16 is addressed. Also not all things are up to present version of FidoCadJ. That's why version 0. Its some kind of preview. I know that you probably don't speak or read Dutch, but you can evaluate the build,  way  of presenting, if  you agree with the examples shown and you can even try some picture references which I fixed. I'm going to find out if I can do the same for tables, for now that's a no go in Latex. Another reason for version 0 is that I didn't have everything,  so I could not set Babel right with Dutch. I did download and installed the necessary files, but setting the file strait is something else  if I have to believe the comments on the net about this subject. So please get in contact.

JoopN, 2023-05-13:
Hello Davide,

Please go to the issues page in Github.
Manual translated to Dutch, version 0.
Want to know how to proceed. One of the things is that I want that you
see it end walk through the document, even if you don't read or speak
Dutch you can see the build, the way its done. Issue 16 is addressed,
have to fix a problem with Babel in order to translate everything
like words like "chapter". Not all is up to date with present version
of FidoCadJ ,so I want to address that too. I have to look after the
pictures, some might need an upgrade or to Dutch or to present
version of FidoCadJ. If you like it I can do the English manual too, 
so that both manuals have the same topics.

regards,
Joop

JoopN, 2023-05-18:
> Hi @JoopN great work! What you can do is to prepare a pull request only for the Dutch manual
> 
> Cheers, D.
Okay, the only thing I get is something with compare one with the other. But there is nothing to compare and in this case its brand new. No button with upload or so. I understand you have to give info what it is, why, how, changes with the English manual besides the translation and so on. All no problem, but first I have to get that document up. Also complained with Github, but probably they don't see it. Help...

JoopN, 2023-05-20:
I hope you have received the Dutch manual, don't hear anything. I busy with a new translation and with it I fond out that I have to fix some parts in the Dutch manual too. Due to fine arrangements of the float system of Latex ;-( . Found also some typo's in part 1 and some words which seems forgotten (Thank you Google ;-(). But after all, its open source,  soI fix it. No, I'm not busy with the English one, you want that I wait with that... (for some reason?)

DarwinNE, 2023-05-20:
Hello,
   I haven't seen any pull request. Did you send me the manual with another way?
If you have done it, the pull request is the way you can show code that I can possibly correct. Do not send anything by email, for instance.
I will not include anything in the repository without a pull request.
Cheers,
D.

JoopN, 2023-05-20:
No, for my try with pull request  see above. Did  send the manual with the answer on an email from you. So that's gone?
As before,  ull  request  seems not to  work for me. Github is shortsighted with only Windows and Mac. I only  get that  I can compare a with b, nothing which tells me how to upload. That's what I get with the option pull request. So help..

JoopN, 2023-05-20:
Here is the pdf, I want that you look at it first, say what you think about it. 
[manual_nl.pdf](https://github.com/DarwinNE/FidoCadJ/files/11522784/manual_nl.pdf)

DarwinNE, 2023-05-21:
It looks very nice!

JoopN, 2023-05-21:
I found a solution for text and pictures in the part about the pcb, this works too with the Dutch translation,  but have  to upgrade it,  now busy with the German translation. I also made a repository, but  still busy with security. Is it possible to do a  pull reguest from a hidden one?

DarwinNE, 2023-05-21:
I don't know. I think you must fork of the FidoCadJ project.
I think this page should give you the instructions: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

DarwinNE, 2023-05-24:
Good point. Leave me a day or two and I'll try to check.

JoopN, 2023-05-24:
In the explanation of "Natural cubic spline" is a reference to the html page http://www.cse.unsw.edu.au/~lambert/splines/natcubic.html. I checked, but this site will give a 404 error, which means that the site is no longer there. For now I removed the reference until you can point me to a new location.

DarwinNE, 2023-05-24:
I checked the code. The maximum number of tokens recognized in a line of FidoCadJ is 10000: https://github.com/DarwinNE/FidoCadJ/blob/ec920a1a22b004ffaca2fc8abacb0bf80da731dd/src/net/sourceforge/fidocadj/circuit/controllers/ParserActions.java#L48
Therefore, for a spline having the following form: 
~~~~
CV aa x1 y1 x2 y2 ... l 
CP aa x1 y1 x2 y2 ... l
~~~~
you have to count (10000-3)/2 that rounded gives 4998 vertex.

Unfortunately, it is true that the page about the natural splines has vanished. Maybe a workaround is to use the Internet Wayback Machine:

https://web.archive.org/web/20141224163714/http://www.cse.unsw.edu.au/~lambert/splines

I remember I contacted the author to ask for permission to use his code, that he granted me.

JoopN, 2023-05-25:
> Unfortunately, it is true that the page about the natural splines has vanished. Maybe a workaround is to use the Internet Wayback Machine: [...]
Thank you for the number of  tokens. 
The website on archive.org  is a nice menu, but all pages are empty and the pointer to the original website does not work due to the 404 error. So  in the end we are empty handed.

DarwinNE, 2023-05-25:
I see, but I do not have any alternative solution... It's a big problem to reference websites that disappear :-(

JoopN, 2023-05-25:
This is the same man https://pclambert.net/interactivegraphs/  ?

DarwinNE, 2023-05-25:
Unfortunately I don't think so. From what I recall it was a computer graphics course.

JoopN, 2023-05-30:
With export to pdf you are limited to Helvetica, Courier, Times of Symbol fonts. These are Windows fonts. So on an other system not running under Windows those fonts will have a different name. The question is how strict those names are? Next how does the export decides if it is the right font, has the font name have to begin with one of those four names or can it also somewhere in the middle or the end of the name?

JoopN, 2023-05-30:
How do you want to have the pull request? All files separate or for each language one zip file?

DarwinNE, 2023-05-30:
There is no need for a zip file. Read this document if in doubt about how to proceed:

https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

Create a subdirectory for each additional language. Put all files related to that language there. Make sort that all the files can be compiled directly from the subdirectory. Include the pdf file in the commits.

JoopN, 2023-06-03:
I have 3 manuals (English, Dutch and German), one uploaded. I'm pushing to left and to right, up and down, made it private, made it public, followed the guidelines, but in the end nothing. Seems you are not open for it. I give up. Github  is not for me. Something which should be easy seems to be an terrible headache. For some reason Github don't want any  code, so you make it as difficult as possible. I can spend my time better.

DarwinNE, 2023-06-03:
GitHub works with a philosophy that works well for large participative project.

However, I checked and I don't see any public repository in your account. Of course you will not be able to commit changes directly to my repository, you have to create your own. Usually things work as follows:
1 - You fork the repository in your account. It is quite easy to do, if I recall correctly there is an icon "Fork" in the top right corner of the DarwinNE/FidoCadJ page. That will create in your account a public repository that will be called JooN/FidoCadJ. You will have write access to this repository.
2 - You commit the manuals and the changes to your repository.
3 - Once you have the commits in your local copy of FidoCadJ, you can create a pull request towards my repository. The system will compare the two repository and propose a pull request.

If you want to have a look at the past pull requests that were merged into the FidoCadJ project, have a look here: https://github.com/DarwinNE/FidoCadJ/pulls?q=is%3Apr

JoopN, 2023-06-04:
Thanks, that helped. Now I'm at the stage that I can create a pull-request, but I can't choose where they have to go to, ie  DarwinNE/FidoCadJ/manuals. Second I have made a /main, in it /Dutch-translation, in it /images and in that /icons. If you start the tex from /Dutch-translation (if that is possible) it should compile because all is at place ie /images and /images/icons are a sub-directory of /Dutch-translation. Or just do the pull-request and you do your own placing into the structure? Need advice in this. For now just the Dutch part, if that's going okay then I can upload English (update) and the German manuals which have there own images and icons directories. The directory /images contain nationalized pictures, the directory /icons is just the standard you already have.

DarwinNE, 2023-06-04:
Ok, now I see the repository. GitHub says it is up to date with the main repository that is mine.
You should learn how to use Git now.
1 - The idea is that you must have a local working directory on your disk (you obtain it with git clone and the appropriate address, click on your repository on the green button "Code" and use that address. It should be something like "https://github.com/JoopN/FidoCadJ.git"
2 - You can modify the local working copy
3 - You stage and commit the changes. Look at the Git documentation. (commands "git add" and "git commit -m")
4 - You push the changes on your project on GitHub. You may need a token that you can obtain from GitHub.
5 - From your updated GitHub repository, you generate a pull request.

JoopN, 2023-06-04:
No, that`s not  it. I can make a pull-request, but you get it all in the main branch and not in the directory /manual. I think you don't want it in the main branch. So or you set it right in your structure ie "/manual/Dutch-translation" and "/manual/Dutch-translation/images" and "/manual/Dutch-translation/images/icons" or I have to do something which doesn't go. I uploaded with this setup, but it seems of no value with the pull-request or it doesn't show right. All is going to your main branch at present state.

DarwinNE, 2023-06-04:
You should just move the files you created under manual/, create the subdirectory where the images should go, make sure everything compiles correctly.
You can work in the main branch, there is no need to create another branch for that.

DarwinNE, 2023-06-05:
I do not understand, maybe there is a certain confusion between directories and branches. Those are two different things. First of all, you will always work on your local copy, then you stage, commit and push the changes to JoopN/FidoCadJ. Make sure you include the files in the right directory and subdirectory.

Then there are the Git branches. You can create a separate branch, or work on the master branch as you wish.

In a second moment you can do the pull request. For small modifications (or less small ones but that do not alter much the core code of FidoCadJ) it is OK to merge in the master branch. It will be my responsibility to review it to accept it or not. I may either ask you to modify something, or I will do the modifications myself, depending what requires less effort.

DarwinNE, 2023-06-07:
Hello,
  I saw pull requests #230 #231 #232, but I haven't had the time to review them. Did you close them? Do you need more information? Even if the files are not in the right spot, I can move them.

DarwinNE, 2023-06-07:
Yes, you should keep the pull request open if you want me to merge it.
I would really like to move on with this project

Please proceed as follows:
1 - Put all the files manual_nl.tex and manuals_nl.pdf related to the Dutch translation of the manual in the "manuals" directory.
2 - Create a directory with a subdirectory "manual/dutch/pic" and put there all the pictures relevant to the Dutch translation only.
3 - Update the compile.sh file.
4 - Create a new pull request.
5 - Leave me some time (at least two weeks) so that I can review your contribution. I may ask you to correct something or I may choose to correct it myself if it is easier.

When the pull request related to the Dutch translation will be merged, you will do the same for the German translation. A not up-to-date French and Italian translation are available, but please do not touch them if you do not speak those languages.

For the update to the English version we are going to discuss later.

Please try to follow the order I described.

JoopN, 2023-06-07:
> Yes, you should keep the pull request open if you want me to merge it. I would really like to move on with this project

I re-opened the three pull requests

> Please proceed as follows: 1 - Put all the files manual_nl.tex and manuals_nl.pdf related to the Dutch translation of the manual in the "manuals" directory. 2 - Create a directory with a subdirectory "manual/dutch/pic" and put there all the pictures relevant to the Dutch translation only. 

Tried this before, don't have that choice.  

>3 - Update the compile.sh file. 
Not sure what  you mean, but  the file is  setup to compile manual-nl.

>A not up-to-date French and Italian translation are available, but please do not touch them if you do not speak those languages.

Had no  intention to do that, my proposal  was different. 
> 
> For the update to the English version we are going to discuss later.

Updated English manual is  the old manual plus additions, nothing changed in old  text except for comments with one picture and one site which is not available anymore. Also pictures are updated to the same degree as in the other manuals. So,don't worry its totally different, its not.   
>

JoopN, 2023-06-11:
I  have uploaded the German manual (3 files) strait in the manual directory (it was open this time). I made before the upload of the pictures a structure "German/pic" and upload the 42 pictures. I don't know what to do with the icons, you didn't use them with the Dutch translation as far as I can see. They are the same as in the English manual. But in the preamb the directory /images and /images/icons is used. I did setup this way and it will compile. The present structure is  different. If you have time left you couldmay be  change  the text

DarwinNE, 2023-06-18:
I accepted the pull request #233 , thank you very much for the German translation! I have some questions, I will ask them in #227. You can cross-link to a particular issue from the pull request using the # character followed by the corresponding number. You can also use a more descriptive title for the pull request (that somewhat can describe the commits it contains). It will be easier to refer to it in the future.

For the Dutch manual, I only moved all the files, I do not have yet LaTeX installed on the computer I am currently using (I changed it recently) so I could not try to compile it. It should refer the Dutch pictures, of course in the directory /manual/netherlands/pic. 

For the English manual, do not use the zip file. Just edit it, commit the changes, push them on the GitHub server and do another pull request. I will be able to see quite precisely all the changes you do. This is the advantage of version control software such as Git and GitHub will allow me to review the differences in the pull request. I may comment directly on the pull request and ask you to do some changes. It is quite practical.

JoopN, 2023-06-21:
I have tried to download .tex files or pictures, but Github has changed the way it works. The downloaded .tex files are HTML files to some degree, but not .tex and certainly not the way its setup. This means you can't process them. Moreover, Latex is a tool for publishing scientific stuff and is by no means a dtp program by far. So the small changes in the manuals for getting the pictures with the text are somewhat lost and every time YOU have to agree with the changes. Editing offline in a dedicated tex editor is not possible.   
Said that, no your changes for the  pictures won't work. In table 2.1 you use icons and those are also in the translated manuals, no difference there. This means that its a graphical thing, so tex uses the rule in the preamble and in table 2.1 you set it to /icons/icon file. So to be successful you need to have an /icons directory with the icon files in the /pic directory. 
As Git changed the way it works I think you can at best set the pdf for direct access, it will open in your browser and from there you can download it. This however doesn't work with tex-file. You have to edit it online, but that is way to slow and does have the above problems, you have to agree. But dtp doesn't work that way. So I think its best to make a zip-file for one language, which can be download as a whole, with the tex-file, may be a readme file and directories with the pictures in the right way.  That is if you like that other people are willing to make some progress with the program, in present way I doubt that very much.  May be this works with Windows or Apple (Github have some manuals about the two), everything else you are on your own. So again 3 days lost for something which could be done in 3 minutes when it was a zip.

DarwinNE, 2023-06-23:
Hello,
   I'm sorry for the difficulties you are experiencing. I have not fully understood some of the issues you have described. If I can give a suggestion, try to create a dummy project on your own and do some experiences to really understand how Git and Github work. Git and in general version control systems are extremely useful tools when you have to maintain a project with more than 100 lines of code. I am globally very happy to have spent some time learning their basis as they are useful even in my daily work (not related to programming).

I just installed LaTeX on this computer, I will correct all the issues with the Dutch version in the next few days so that the file will compile. I'll let you know.

DarwinNE, 2023-06-24:
I corrected the references to the icons. Now the file correctly compiles.
Can you please check that everythins is ok?

JoopN, 2023-06-30:
I downloaded everything with "Code" and then download zip file. Copied Dutch to a directory and compiled it with TexStudio. No errors, checked with a manual from before update and all looks the same. So Dutch is going okay. 
Not so with the German. The way how you want to do it does not work that way. The TEX file has no knowledge of what your root is. It works from the directory in which the tex file resides. So all references in the preambule has as root this directory. Going to /images/pic etc will not work because TEX will translate that to /german/pic/images/pic etc. Looks to me you have to setup same way as you did with Dutch, ie convert the ico's to png's and set them in the directory /german/pic.

DarwinNE, 2023-06-30:
If the Dutch translation is complete, I close this issue.

closed 2023-06-30 by DarwinNE