FidoCadJ/FidoCadJ #213: Use of a different language in FidoCadJ then the locale

JoopN, 2022-05-24:
When I start FidoCadJ then it will come up in Dutch because my computer is in Dutch, called the locale. But sometimes you want to  work in an other language, there could be different reasons. Mine is that I'm busy with translating, but may be you want to share a screenshot with a friend in an other country. I did not find a way to switch the language, I was thinking about something on the commandline as options.

DarwinNE, 2022-05-25:
It is possible, with the command line option -l, followed by the two letter code of the locale you want to use.
For instance:

~~~~
java -jar fidocadj.jar -l en
~~~~

should start FidoCadJ with the English user interface, even if your locale is Dutch.

DarwinNE, 2022-05-26:
Command line options are described (albeit briefly) in the English manual. Look at paragraph 2.9, page 24 of the current version of the manual.
However, you can obtain a list of the options available by means of the `-h` option, look at page 26 for an example (I think it is reasonably updated).

JoopN, 2022-05-26:
Found it, table 2.2 sits not behind 2.8 but somewhere in the mids of 2.9, this is not logic, may that's why I missed it because when you start reading 2.9 it stops without examples, but those are behind table 2.2 and I thought it was something different. Have to fix that, but right now I don't have any idea what will happen if I make a pdf, have to find out how that works and I  have to install some add-on's into Texstudio.

DarwinNE, 2022-06-07:
This is a result of the fact that table 2.2 (as all figures in the manual, by the way) is a floating object. I can move it around slightly, but ultimately LaTeX decides where to put it so to optimize the composition of the page. I am not sure where it would be best to put it, by the way. It sounds logical to put it close to the line where the table is referenced, in the paragraph 2.8. Since the table has to occupy a full page, LaTeX has put it in the next one.

DarwinNE, 2023-05-19:
Ok!

closed 2023-05-19 by DarwinNE