FidoCadJ/FidoCadJ #218: ruler and popup menu are the same key with a pen (and tablet)

JoopN, 2022-06-21:
When I use an ordinary mouse I will develop RSI within two weeks or so. Therefore I bought a pen and tablet from Wacom and use  them as a mouse, RSI gone. So I have them on every computer, all mouse are gone. In most times the right click is not a problem, the  pen functions can be programmed in the driver, so no problems. But in FidoCadJ two functions are fighting which can go first. Or I  do get the ruler or I get the pop up menu with functions like parameter field and so on. I think  it  would be an option to  make an other  button or may be more buttons in which you can choose or ruler, or pop up or (what I have still to find out ;-)). Sorry, this happens when you are going to translate, sometimes you have to test to get the right idea and the right translation.

DarwinNE, 2022-06-25:
IIRC you get a ruler when you right-click and drag.
May it happen that with the pen you right-click and move a little the cursor? In this case, the event is recognized as a right-click and drag instead of a simple right-click.
BTW, I implemented the ruler function many years ago when I had a G5 iMac with a mouse. I never use the ruler as I only exploit a trackpad now and I no longer use a mouse.

JoopN, 2022-06-25:
The tip of the pen is the left click or mouse button 1, the right click is programmed to the second button, mouse button 2. Mouse button 3 has no function because its a very difficult to reach it and hold you tip still. There are the next combinations. Hover and use second button, or rest on the tip and therefore use left click or mouse button 1. If I draw a line then its sometimes difficult to choose the line. May be its me that you react a little different day by day. Sometimes its very difficult to get the popup menu and you get always a green lined box with figures. Or you get the popup menu straight away. If it doesn't go I enlarge the drawing so everything is a little bigger and more easy to choose.

DarwinNE, 2022-06-26:
I think the right button + drag was not well chosen in the first place. It is not a common gesture and it is almost impossible to do with a trackpad or an alternative pointing device.
What shall be used instead? A key+click and draw? Which key so that it is something that people may found intuitive?

DarwinNE, 2022-07-03:
Maybe the best trade-off would be to associate click and draw with a keyboard button.

JoopN, 2022-07-03:
I find out that its more the problem of catching the point with the pen. Might be that my pen tip sees more then a mouse pointer. Sometimes straight away the right menu, sometimes the green square keeps coming. For me I don't move the pen, but when I want to set the point to 100.100 with "snap to grid" off then I see the figures going up and down with one or more pixels. I have to push the button on the pen, so tiny movements are there.

DarwinNE, 2022-07-04:
I think I see what is going on. I will check.