FidoCadJ/FidoCadJ #220: Double click on fcd file cause an open of FidoCadJ but on a new document

gremirarunico, 2022-07-05:
When I double click a fcd file on finder FidoCadJ correctly load, but it doesn't load any schematic. Even if I drag and drop the schematic on the windows it won't be loaded.
I have to do File => Open and select the file to be correctly opened, or alternatevly copy paste the text contained in fcd.

Info:
FidocadJ version 0.24.8
OS MacOS 12.4 (21F79)

DarwinNE, 2022-07-05:
I confirm there is a bug on the macOS version. I will have a look at it.

giancarlo-deviti, 2023-11-12:
Hi, I confirm the problem reported from gremirarunico.
Info:
FidoCadJ version: 0.24.8
Mac OS version: 14.1.1 (Sonoma)

DarwinNE, 2023-11-12:
Thank you! I implemented the behaviour a few years back and it worked back then. Unfortunately something broke during the Java and macOS upgrades. I'm quite busy at the moment, if someone wants to have a look, I'm not against it!

manufino, 2024-08-21:
@DarwinNE, can you work on this? I don't have a Mac.

DarwinNE, 2024-08-21:
Yes, of course. I will have a look at it in a few days.

DarwinNE, 2024-11-26:
I tried to use this bundle Info.plist with some success:

~~~~
<?xml version="1.0" ?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "https://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
 <dict>
  <key>LSMinimumSystemVersion</key>
  <string>10.9</string>
  <key>CFBundleDevelopmentRegion</key>
  <string>English</string>
  <key>CFBundleAllowMixedLocalizations</key>
  <true/>
  <key>CFBundleExecutable</key>
  <string>FidoCadJ</string>
  <key>CFBundleIconFile</key>
  <string>FidoCadJ.icns</string>
  <key>CFBundleIdentifier</key>
  <string>net.sourceforge.fidocadj</string>
  <key>CFBundleInfoDictionaryVersion</key>
  <string>6.0</string>
  <key>CFBundleDocumentTypes</key>
	<array>
		<dict>
			<key>CFBundleTypeExtensions</key>
			<array>
				<string>fcd</string>
			</array>
			<key>CFBundleTypeIconFile</key>
			<string>generic.icns</string>
			<key>CFBundleTypeName</key>
			<string>FidoCadJ file</string>
			<key>CFBundleTypeRole</key>
			<string>Editor</string>
			<key>LSItemContentTypes</key>
			<array>
				<string>net.sourceforge.fidocadj.file</string>
			</array>
		</dict>
	</array>
  <key>CFBundleName</key>
  <string>FidoCadJ</string>
  <key>CFBundlePackageType</key>
  <string>APPL</string>
  <key>CFBundleShortVersionString</key>
  <string>0.24.8</string>
  <key>CFBundleSignature</key>
  <string>????</string>
  <!-- See https://developer.apple.com/app-store/categories/ for list of AppStore categories -->
  <key>LSApplicationCategoryType</key>
  <string>Unknown</string>
  <key>CFBundleVersion</key>
  <string>100</string>
  <key>NSHumanReadableCopyright</key>
  <string>Davide Bucci and many others</string>
  <key>NSHighResolutionCapable</key>
  <string>true</string>
 </dict>
</plist>
~~~~

The mechanism behind `CFBundleTypeExtensions` seems to work fine (VLC still uses it) and it seems to work.
I'm updating the bundle of the macOS version.

DarwinNE, 2024-11-26:
I changed a few details (and I added the file icon to the bundle).
I just updated the file in the macOS release so that the application comes with the new Info.plist.

https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.8

If someone confirms that the problem is solved, I will close this issue.