FidoCadJ/FidoCadJ #221: Selecting elements

JoopN, 2022-07-27:
When you go big in drawing and use something like the millimetre grid some strange things happen.
Selecting with snap to grid is impossible when the element is not on a grid line (say settings are 5mm). Switching snap to grid off does help, but next there seems some strange things when selecting. I have drawn a light tube, both sides black line with on grid line a yellow line. See example. It looks like there is some room around that yellow line on grid line which sees to it that you always select that element. The element 1mm above or below is very difficult to select. You have to go to 1000% or something like that.
Example light tube (ADAT light pipe)
LI 112 119 102 119 0
LI 102 120 112 120 4
LI 112 121 102 121 0

Same for multiple selections, I had drawn lines (connections between different block elements) and because it is not clear what line it is somewhere in the drawing I had added text to the line for what it is (like TY 181 935 2 1 0 0 2 Liberation++Sans++Narrow VU/PPM IN L). So you can follow that line through your drawing all the way to the other end. Sometimes I have four or more lines next to each other and I had to move the text a grid block (5mm) up or down or left or right. I had great difficulty in selecting more than 2 or 3 text lines. Sometimes the connection line was selected instead of the text line next to it. Again like  there  was  some room around the "connection line".

Most strange observations in this act is that somewhere else in the drawing something is selected, erases, moved or some line has changed colour. And that all happen in a part of the drawing which is even not on screen.

Also it is getting difficult to move around and to insert a lib item, but this could be the problem that I entered the border from my Raspberry Pi. 

Said this all, its not all negative, made my own libs and draw complex studio setups with different equipment. Your program did give me good insight what to do and what I needed. Plan is possible, now the part how get it all with present chip shortage, that  is another challenge. So I'm not there yet, but one hurdle is taken.

Regards,
Joop

DarwinNE, 2022-09-02:
Hello,
   I tried to replicate the behaviour you describe with the light tube code, but I'm not sure I understand what you mean.
I have a 1000% zoom and if I click on the black lines I can select them with no problem.
The selection rationale is quite complex and may behave strangely in some situations, but I haven't seen nothing strange in the example you provide.
On the other hand, I confirm that if there is a handle nearby, FidoCadJ tries to prioritize it.
When there are two drawing elements very close to the point where you click, it is also possible that the prioritization changes depending on the layer on which they are placed.

JoopN, 2023-05-18:
Did a test, gave the black lines not layer 0, but layer 15. Now you can't select the line with color 4. I have to click next  to the line somewhere in the free space in order to select a black line when it was on layer 0. May be its the handle. On the other hand, what I do is extreme.

DarwinNE, 2023-05-19:
Hello,
  indeed it is problematic to combine the layer priority rules with the rationale used to select objects.
I can think to solutions that only work in some particular cases, but it is difficult to have something that works perfectly for all situations...

Cheers,
D.

DarwinNE, 2024-09-01:
I think we can close this issue.

closed 2024-09-01 by DarwinNE