FidoCadJ/FidoCadJ #227: German translation of the manual

JoopN, 2023-05-20:
I'm busy with it, somewhere in chapter 2, lets see how far I can get. I use Google for the translation and have sometimes correct that translation, but its amazingly fast with "rough" translation, can't type that fast, no way. So lots of copy/past and read what I have, correct if necessary.

DarwinNE, 2023-05-20:
I have anything against an automatic translation, but please double check that the quality of the translation is good nonetheless.

JoopN, 2023-05-20:
> I have anything against an automatic translation, but please double check that the quality of the translation is good nonetheless.

Don't  worry, I  do the best  I can. Everything I do read, and when in doubt I try to translate it in an other way in order to see if that gives the same result. Besides translation, you also have to deal with how you translate it, jargon or not?. sometimes difficult.

JoopN, 2023-05-24:
I'm somewhere at 3.3, I hope I can finish the translation this week.

JoopN, 2023-05-25:
I have a complete version ready, I will read it after some days in order to see if I see strange translations or compiling errors. In between I will see what I did differently with the original English version and make notes of those changes in the TEX file.

DarwinNE, 2023-06-18:
I accepted pull request #233 with the German translation of the manual, that's great, thank you! 
All the pictures were in /manual/ and not in /manual/german/pic and I moved them in the subdirectory. I specified the correct directory in the German manual. If you can, try to compile it and correct the issues.

DarwinNE, 2023-06-25:
I checked the references to the icons in the manual. Now it correctly compiles. The translation is complete, I close this issue. I may be open it again if something will need to be translated in the future (i.e. new paragraphs in the manual, etc...).

JoopN, 2023-06-30:
No, it does not compile, see my comment with Dutch

DarwinNE, 2023-06-30:
Ok, I reopen the issue and I'll have a look in the next few days.

DarwinNE, 2023-06-30:
I copy paste from #212, so that it is easy to follow the discussion:

>Not so with the German. The way how you want to do it does not work that way. The TEX file has no knowledge of what your root is. It works from the directory in which the tex file resides. So all references in the preambule has as root this directory. Going to /images/pic etc will not work because TEX will translate that to /german/pic/images/pic etc. Looks to me you have to setup same way as you did with Dutch, ie convert the ico's to png's and set them in the directory /german/pic.

That's strange, I tried again and I could correctly compile the manual in German. I tried before and I checked again this evening.

The file `manual_de.tex` is in the manual subdirectory of the FidoCadJ project and contains `\graphicspath{{german/pic}}` that of course refers to the place where the LaTeX source is present. All the pictures, including the icons are in the  subdirectory `manual/german/pic`.

Are you sure you are compiling the up-to date version of the manual_de.tex file?

JoopN, 2023-07-05:
Downloaded again and its still the same. But I think I do know why. You do the job within GIT, I download (like everyone) all and do the job in a TEX editor. Then it fails. 
I also don’t understand why you have three different setups. When I do something like that, all will be the same and oriented to the end users capability.
 
With the English manual you have images/ and for the icons in table 2.1 icons/ which will translate to images/icons/. 
With the Dutch manual you have netherlands/pic/ and the icons in table 2.1 are placed in the same directory. 
With the German manual you have german/pic and for the icons in table 2.1 images/icons/ which will translate if not started with GIT to german/pic/images/icons/. 

I think that the solution presented with the Dutch manual is the way to go, pictures and icons in same directory. 

At present with the given solutions in the TEX files the German will not compile if started with a TEX editor. Its your program, your manuals, up to you. But if you want helping hands then you have to make it as simple as possible and you think GIT is simple, may be for the given programs by GIT for Windows and Mac, the rest is on its own and GIT is far from simple. So I learned myself TEXStudio for editing TEX files. That’s not going through GIT but who knows what’s inside this program. For now I delivered a Dutch and a German translation. Busy with the English for pointing out what I have done in order to get manual with same appearance as the Dutch and the German manuals.

DarwinNE, 2023-07-06:
> Downloaded again and its still the same. But I think I do know why. You do the job within GIT, I download (like everyone) all and do the job in a TEX editor. 

I strongly suggest to use Git. Learning how to use versioning software is without doubt one of the best investments I did in computers in the last twenty years. They are tools you don't realize you need until you realize how powerful they are. They have their quirks, of course, but it is definitely worth using them. A difficulty is the jargon. For instance, words like fork, commit, branch (and others) have a definite meaning in the context.

That being said, you can proceed as follows:

1 - You synchronize Joop/FidoCadJ on GitHub with DarwinNE/FidoCadJ (in this moment your fork is 20 commits behind). If you look to https://github.com/JoopN/FidoCadJ while being logged, you have a button that allows to do this called "Sync fork".
2 - If you use Git from the command line, you go to your repository and type `git pull`

Those two operations allow you to have a copy that is exactly in sync with mine, no matter which changes I have done (and without to have to download again all the project).
If you want to change something you do it on the repository on your computer and then you type:
`git add -u`
`git commit -m "This is a description of the change"`
`git push`

and then you create a pull request on GitHub. It's an incredibly powerful and flexible method when tracking the changes is not trivial.

> Then it fails. I also don’t understand why you have three different setups. When I do something like that, all will be the same and oriented to the end users capability.

The simple reason is that I wrote the first draft of the manual in 2008 and I don't recall all the details.

> With the English manual you have images/ and for the icons in table 2.1 icons/ which will translate to images/icons/. With the Dutch manual you have netherlands/pic/ and the icons in table 2.1 are [...]

The result should be the same no matter if the project is handled with git or with another technique. Of course, the files should be in the same place. But I got the idea and I do not understand why the manual compiles in my case. I tried to correct that in the commit [650bd48](https://github.com/DarwinNE/FidoCadJ/commit/650bd48778697cc81b33ef80993b8ab144cc85f2)
 
> I think that the solution presented with the Dutch manual is the way to go, pictures and icons in same directory.

I agree. I tried to address that in my last commit. Let me know if it is ok now.

> At present with the given solutions in the TEX files the German will not compile if started with a TEX editor. Its your program, your manuals, up to you. But if you want helping hands then you have [...]

The real problem is that collaborating on a nontrivial project is an intrinsically complicated task. I do not know any better approach than using a versioning tool, unfortunately. GitHub (or Sourceforge) are practically built around versioning software. FidoCadJ is a small project, but there have been about 20 contributors in the last ten years. If you consider me able to keep track of all the modifications without an appropriate tool, you are greatly overestimating my capabilities :-)

My suggestion is to learn some basics of Git and use the editor that you prefer, the two programs can be kept separate. The responsibility of Git and GitHub will be to keep your copy in sync with mine and propose modifications as pull requests. The responsibility of your TeX editor will be to... edit the LaTeX files.

JoopN, 2023-07-28:
I had some time at the beginning of this year, after that I had to give trainings for how to do a job at my former work for three months.  So my agenda has shifted three months into the future. In short, I don't have the time right now to learn a new program and understands it ins and outs. I did overhaul the English manual and put in comments about what I did. It still compiles right and does have a layout which is similar to the German and Dutch manuals. But there are subtle differences due to the fact that in whole English uses less space.

DarwinNE, 2023-07-29:
Hello,
   no problem!
If you prefer, put the updated english manual in a different file or directory and do a pull request, I will rename or move the file in the appropriate place.
Kind regards,
D.

closed 2023-08-30 by JoopN