FidoCadJ/FidoCadJ #228: Symbol definition dialog

JoopN, 2023-05-20:
Enclosed  a picture with translated text. The room behind the checkbox "snap origin on grid" is too short for some other languages. The translation is correct, but  there is not enough room for the text. Seems to me an easy fix.

[image]

DarwinNE, 2023-05-24:
I see. I'll have a look at this soon.

DarwinNE, 2024-08-12:
@manufino has worked on this issue. Can we close it, or there are other changes to be done?

manufino, 2024-08-13:
I think we can close this. This should have resolved the issue.

[image]

JoopN, 2024-08-14:
Op Wed, 14 Aug 2024 12:37:22 -0700
Davide Bucci ***@***.***> schreef:

> @JoopN, did the last changes improve the issue?
> 

Sorry, did not know that there was a new version. Have to download
and see.

Regards,
Joop

manufino, 2024-08-14:
> Op Wed, 14 Aug 2024 12:37:22 -0700 Davide Bucci ***@***.***> schreef:
> @JoopN, did the last changes improve the issue?
> Sorry, did not know that there was a new version. Have to download and see. Regards, Joop

I think you need to download the source code and compile it.

manufino, 2024-08-22:
@DarwinNE, You can close this one for me as well.

closed 2024-08-22 by DarwinNE