FidoCadJ/FidoCadJ #234: Dialog Text parameters

JoopN, 2023-08-30:
I make a schematic drawing with lots of parts. When I'm finished I'm going to give the parts a name like R1, T1 and so on. After that I do the values or part numbers like 220k or LM386, something like that. And after that I put some calculations in the schematic in explaining how I got there with that particular value. I do this with different layers, read colors. Blue for component id, red for value, green for calculations, light blue for calculated value's or measured value's.
So far nothing new. I can set font in Options, I can set the layer=color in the workspace bar. But there are more options in the parameter field from Text found in the toolbar. For every entry I have to adjust X size, Y size, Angle, Mirrored, Italic and Boldface. It would be nice if I had an option to set those "fixed" until I update the fields next time. In this way I can do the above much faster. I already copy same values to other components,  but still for a lot of components I have to do all over and set the named six every time again, as blank already exists it is mostly narrowed down to X size, Y size and Boldface.

manufino, 2024-08-18:
@JoopN, An idea I've had for a while was to be able to change the parameters of multiple selected identical primitives. This would solve this problem from what I understand. 
In the current state, if you select multiple primitives, you can only change the layer (even if the primitives are of the same type). It would be nice if you could also change other parameters.

JoopN, 2024-08-19:
Op Sat, 17 Aug 2024 18:24:32 -0700
Manuel Finessi ***@***.***> schreef:

> @JoopN, An idea I've had for a while was to be able to change the
> parameters of multiple selected identical primitives. This would
> solve this problem from what I understand. In the current state, if [...]
@Manuel,

That's right, I have a drawing with lots of text alongside lines to
point out where its coming from or where its going to. In the
printing process it turned out that even printed on A2 it was
to small and not readable. So I had to change size for a few
hundred lines of text. As the drawing is readable text I solved
the problem with an ascii text editor with find and replace. Sorry for
that. And there are other things you want to change for a group,
part bold or italic, other font, that kind of things. For me its
problems with text, drawings not so.

manufino, 2024-08-21:
@DarwinNE, @JoopN  , I'm working on this feature, currently limited to `PrimitiveAdvText` only.Do you think it would be useful to allow changing the text string for all selected ones (in addition to other parameters) ?

DarwinNE, 2024-08-22:
Shall we close this issue?

manufino, 2024-08-22:
I think we can close this; what was requested has been implemented.

closed 2024-08-22 by DarwinNE

JoopN, 2024-08-23:
Op Wed, 21 Aug 2024 12:04:37 -0700
Manuel Finessi ***@***.***> schreef:

> @DarwinNE, @JoopN  , I'm working on this feature, currently limited
> to `PrimitiveAdvText` only.Do you think it would be useful to allow
> changing the text string for all selected ones (in addition to other
> parameters) ?
> 

The text string itself is very object specific in my drawings, but
may be others find it useful. But everything else would be nice,like
layer, X size, Y size, Angle, Mirrored, Italic, Boldface and Font.