FidoCadJ/FidoCadJ #236: Distinctive selection in AutoCAD style

manufino, 2024-08-13:
I have already implemented this feature, also improving the selection system for primitives, which can now be selected by grabbing even a small part of them, not just through the handles.

Here you can find documentation regarding this:
https://help.autodesk.com/view/ACD/2023/ENU/?guid=GUID-243E4DD0-8947-4905-AFE2-BE9B903A8C3F

I had to implement an "intersects" method on all primitives to perform an accurate selection process.
The most challenging were the "[PrimitiveComplexCurve](https://github.com/manufino/FidoCadJ_NetBeans/blob/3d4e50601eb3bbf8451eab3159ce2fd028285c23/src/fidocadj/primitives/PrimitiveComplexCurve.java#L1188)" and "[PrimitiveBezier](https://github.com/manufino/FidoCadJ_NetBeans/blob/3d4e50601eb3bbf8451eab3159ce2fd028285c23/src/fidocadj/primitives/PrimitiveBezier.java#L525)" classes, where I had to implement specific selective algorithms.
This implementation also involved other classes to manage the selection method.

DarwinNE, 2024-08-13:
Hi @manufino,
   it seems a very good idea. Do you want to implement it?

Cheers,
D.

manufino, 2024-08-13:
@DarwinNE, I've already completed and tested it (at least on my end), but I'm stuck with the pull requests due to conflicts.
The conflicts aren't related to the implementations themselves but to the resources that need to be incorporated. In my branch, I've moved them and had to adjust the paths in the code, which is causing an incompatibility between my master branch and the one I created for the pull request, "fidocadj_origin."
I'm trying to figure out how to resolve this so I don't go crazy with every pull request I need to make.

manufino, 2024-08-13:
This implementation is included in my pull request #240

DarwinNE, 2024-08-15:
I am checking the code you added in the recent pull request. Nice work. I noticed it seems that sometimes there are some redrawing issues. Probably the definition of the "dirty" part of the redraw area is not correct in same situations. Here is what I obtain by moving the mouse alternatively from one kind of selection style to another one:

[image]

manufino, 2024-08-15:
Very strange, this doesn't happen on my system.
I can't reproduce the problem on Windows.
Could it be a problem strictly related to macOS?

manufino, 2024-08-15:
Could this be the solution?

on "CircuitPanel":
```java
    /** Determine the direction of the selection.
     
        @param isLeftToRight True if the direction is from left to right..
                             False if it is from right to left.
     */
    public void isLeftToRightSelection(boolean isLeftToRight)
    {
        if(isLeftToRight) {
            this.selectionColor = Color.BLUE;
        } else {
            this.selectionColor = Color.GREEN;
        }
        // Force the redraw when the selection direction changes.
        this.repaint();
    }
```

DarwinNE, 2024-08-15:
It seems to work.
In my experience, the various operating systems have a slightly different behavior related to the repaint. There are just details, but ensuring that the code works correctly on all OSes usually ensure a better overall quality.

You may also see if you can declare only a certain section of the image as "dirty", only the "dirty" region will be repainted. This is usually a faster strategy rather than repainting the whole component. But we will be dealing with the optimizations later (if needed), the important thing is that the code now works.

On another point, I noticed a certain difficulty to handle the spline in this case:
~~~~
[FIDOCAD]
FJC B 0.5
CV 0 30 30 70 20 55 45 110 40 45 10 0
LI 30 15 85 45 0
BE 30 25 70 10 40 40 95 20 0
~~~~

manufino, 2024-08-15:
@DarwinNE, Can you explain the problem you're encountering in more detail?
I tried that draw and I don't notice any issues in my environment..

DarwinNE, 2024-08-15:
Exactly, in many case it is impossible to select correctly the spline.

manufino, 2024-08-15:
Okay, I understand the problem. I'm working on it.
I should be releasing a patch to fix this problem shortly.

manufino, 2024-08-15:
@DarwinNE, Okay, I should have fixed everything. It was a major oversight that prevented proper selection.

DarwinNE, 2024-08-15:
Now it seems to be working as charm, thank you!
I will continue checking in the next few days.

Another situation that is very tricky to handle  is text with index and exponents. 
The original FidoCadJ code is by the way not perfect on this (see #197).

BTW, it would be possible to use the green color for the "integral" selection (that is the "historic" strategy adopted by FidoCadJ) and the blue color for the "partial" selection.

manufino, 2024-08-15:
> Now it seems to be working as charm, thank you!
> I will continue checking in the next few days.

Good.

> Another situation that is very tricky to handle is text with index and exponents.
> The original FidoCadJ code is by the way not perfect on this (see #197).

I'll try to thoroughly check the text handling; let's see if I can find a solution.

> BTW, it would be possible to use the green color for the "integral" selection (that is the "historic" strategy adopted by FidoCadJ) and the blue color for the "partial" selection.

It's already like that. The right-to-left selection is the classic one; it has just been improved. Now it recognizes all parts of the primitives, whereas before, only the vertices were considered. As for the left-to-right selection (the blue one), it selects only and EXCLUSIVELY the primitives that are COMPLETELY contained within the selection rectangle.

DarwinNE, 2024-08-15:
> I'll try to thoroughly check the text handling; let's see if I can find a solution.

It's another tricky problem. FidoCadJ 0.24.8 worked "sufficiently well" with the text, but it was far from perfect. If you like to work on such kinds of problems and you see a way to improve things, do not hesitate.

> It's already like that. The right-to-left selection is the classic one; it has just been improved. Now it recognizes all parts of the primitives, whereas before, only the vertices were considered. [...]

I checked comparing with FidoCadJ 0.24.8 and I think you are right. It makes sense to leave things as you did.
By the way, shall we give a name for the two strategies?
- Inclusive selection
- Completive selection

I'd like to add a small message in the text area to fully discriminate between the two. By the way, if you have terms for an Italian translation, I'm interested, too :-)

manufino, 2024-08-15:
> It's another tricky problem. FidoCadJ 0.24.8 worked "sufficiently well" with the text, but it was far from perfect. If you like to work on such kinds of problems and you see a way to improve things, do not hesitate.

I've been working on this all evening; it's a pretty tricky problem to solve, but I think I can manage it. I'm doing a lot of debugging to figure out how to proceed.

> I checked comparing with FidoCadJ 0.24.8 and I think you are right. It makes sense to leave things as you did. By the way, shall we give a name for the two strategies?
> 
> * Inclusive selection
> * Completive selection

Yes, nice, I like them.

> I'd like to add a small message in the text area to fully discriminate between the two.

Yes, good idea. Can you take care of it?

> By the way, if you have terms for an Italian translation, I'm interested, too :-)

I don't know; I've never been good at communication science ;-)
Maybe a literal translation will suffice ?
- Selezione inclusiva
- Selezione ... _mhmm ... ehm .. You do it; it's better that way._

DarwinNE, 2024-08-16:
> Yes, good idea. Can you take care of it?

With pleasure. I'll work on this between today and tomorrow.

DarwinNE, 2024-08-16:
I added the messages. This required some refactoring around the various controllers, listeners etc.
After a little bit of consideration, I propose to use the following English description:
- "Inclusive selection" to indicate the selection strategy where an element is selected when part of it is included in the selection rectangle: left to right (blue hashed line)
- "Selection of complete elements" to indicate the selection strategy where an element is selected only when it is entirely included in the selection rectangle: right to left (green continuous line)

I hope this jargon would make sense, if there is room for improvement let me know. I translated it in French and Italian and opened the other issues associated to the different translations (hoping that someone can do the work).

manufino, 2024-08-16:
> * "Inclusive selection" to indicate the selection strategy where an element is selected when part of it is included in the selection rectangle: left to right (blue hashed line)
> * "Selection of [...]

@DarwinNE, maybe you meant: 

 * "Inclusive selection" to indicate the selection strategy where an element is selected when part of it is included in the selection rectangle: **right to left** (green continuous line) 
 * "Selection of complete elements" to indicate the selection strategy where an element is selected only when it is entirely included in the selection rectangle: **left to right** (blue hashed line)

DarwinNE, 2024-08-16:
You are right, I corrected.

manufino, 2024-08-22:
@DarwinNE, I've tested everything thoroughly and fixed the last details; this can be closed as well.

closed 2025-12-27 by DarwinNE