FidoCadJ/FidoCadJ #237: Package names refactorig

manufino, 2024-08-13:
On my fork, I performed a complete refactoring of the package names.
In projects like this, it doesn't make sense to apply the Oracle convention, for example:
```java
import fidocadj.dialogs.ParameterDescription;
import fidocadj.dialogs.LayerInfo;
import fidocadj.export.ExportInterface;
import fidocadj.geom.MapCoordinates;
import fidocadj.geom.GeometricDistances;
import fidocadj.globals.Globals;
import fidocadj.layers.LayerDesc;
import fidocadj.graphic.DecoratedText;
import fidocadj.graphic.PointG;
import fidocadj.graphic.DimensionG;
import fidocadj.graphic.GraphicsInterface;
```

It's better than this:

```java
import net.sourceforge.fidocadj.dialogs.ParameterDescription;
import net.sourceforge.fidocadj.dialogs.LayerInfo;
import net.sourceforge.fidocadj.export.ExportInterface;
import net.sourceforge.fidocadj.geom.MapCoordinates;
import net.sourceforge.fidocadj.geom.GeometricDistances;
import net.sourceforge.fidocadj.globals.Globals;
import net.sourceforge.fidocadj.layers.LayerDesc;
import net.sourceforge.fidocadj.graphic.DecoratedText;
import net.sourceforge.fidocadj.graphic.PointG;
import net.sourceforge.fidocadj.graphic.DimensionG;
import net.sourceforge.fidocadj.graphic.GraphicsInterface;
```

This is my opinion; what do you think?

DarwinNE, 2024-08-13:
I fully agree!

manufino, 2024-08-13:
This implementation is included in my pull request #240

manufino, 2024-08-15:
I think this can be closed.

closed 2024-08-15 by DarwinNE