FidoCadJ/FidoCadJ #238: My fork is diverging too much from the original project

manufino, 2024-08-13:
@DarwinNE, I'm opening this issue to discuss how to proceed. I've made substantial changes both to the directory structure and many source files. 
Merging my branch for the pull request is becoming quite challenging.
The most impactful changes are those related to resource management, such as icons, having changed the directory structure, I had to move all the resource files as well to achieve a cleaner project organization.

Could we consider a complete revision of the directory structure, removing parts that are no longer necessary, such as "busy being born" or "OSes"?

DarwinNE, 2024-08-13:
You can propose a refactoring of the directory structure and discuss it here. 
Just take some time to take a broader view with respect to the conventions of a single IDE or development strategy.
For instance, I do not agree that OSes is not useful. I used the "mac" subdirectory relatively recently and I spent a year and a half of my free time developing the (unfortunately aborted) android/fidocadj version.

manufino, 2024-08-13:
One thing I don't like is having resources inside the "bin" folder, such as translation files, icons, and libraries. What do you think about creating a folder in the root directory called "resources," where we can place all the resources to be incorporated into the JAR?

DarwinNE, 2024-08-13:
I like this very much!!! But then the resources must be copied in the bin directory by the build scripts?

manufino, 2024-08-13:
Using scripts could be a solution. I'm trying to gather more information about it since I'm not an expert in the Java ecosystem (my background is mainly in C++ programming).

manufino, 2024-08-15:
For now, I've gathered all the resources in the first level of the bin directory. Check out the latest commits I've made.

DarwinNE, 2024-08-15:
I think we can close this issue, with the pull request #240

closed 2024-08-15 by DarwinNE