FidoCadJ/FidoCadJ #241: Create a new macro to insert images into the drawing

manufino, 2024-08-13:
The idea would be to create a new macro that allows inserting PNG or JPG images into the drawing. The images could be encoded in base64 to embed them directly into the drawing's code. These images should be scalable, rotatable, and movable as desired.

The only problem would be backward compatibility with versions that do not support this feature.

I think it would be a useful feature to implement.

DarwinNE, 2024-08-14:
A difficulty is that this would result in very big files. I never tried to implement this as the original FidoCAD philosophy was to create very small text files. However, times have changed since the year 2000.

Concerning the backward compatibility, this is not an unsolvable problem: after all, the format evolved during the development of FidoCadJ.

manufino, 2024-08-15:
We could set a maximum size limit for the images that can be embedded. Perhaps we could implement a window that allows cropping the image, so you can use only the area you want to embed.

DarwinNE, 2024-08-15:
Choosing a maximum file limit is maybe difficult, as the limit would greatly depend from how the program is used. In certain situations it may be not appropriate.
Maybe a better approach can be to inform the user of the resulting file size while embedding the image.

manufino, 2024-08-15:
> Maybe a better approach can be to inform the user of the resulting file size while embedding the image.

Yes, good idea. Then, as I mentioned, we could provide the option to crop the image if needed, perhaps with different cropping methods (not just rectangular).