FidoCadJ/FidoCadJ #266: Risk of overwriting a macro in a library

manufino, 2024-09-01:
The `DialogSymbolize` generates a key for the selected library by default upon opening. 
If the library is changed, the key is not regenerated, which can lead to overwriting a macro in a library. 

We need to implement a system that regenerates a free key each time the library is changed from the combobox. 
Additionally, it's better to create a check during the macro insertion phase that, if a key already exists, asks the user if they want to replace it.

manufino, 2024-09-01:
@DarwinNE, I believe I've resolved this issue with commit 9b8f5244fd53eba01f0c67f85537ce593a358573. If you can confirm that everything is working correctly now, we can close it.

DarwinNE, 2024-09-01:
Great, thank you. Let's close this one for now.
We are going to release 0.24.9 alpha so that we can get some feedback.

closed 2024-09-01 by DarwinNE